vitormhenrique/OctoPrint-Enclosure #378: Offset AM2302

Mikelomi, 2020-12-03:
Hello, I use your Plugin with many Features, but there is a little thing that i miss.
My Sensor Am3202 works perfectly with Temp but the Humidity alwas about 8-10 percent to much. 

Is there a possabillity to set an Offset corection to the values.  Also would it be nice to have control my Drybox by Humidity Sensor for start /Stop Drying.
. Thank You,for doing this good Job since.

Sorry for my bad English

TazerReloaded, 2020-12-04:
The AM2302 is a digital sensor, performing conversion and compensation internally, so the reported value should be correct. What kind of measurement did you use as a reference?
Regarding the offset feature: This could be benificial for all kinds of sensors, not just the AM2302, and is easy to implement.
The other feature, controlling an output with the humidity reading, is a bit more complicated, because sensor outputs can be expected to fluctuate and that would also make the output unstable. The simplest solution would be to define a value when the output should be switched on, and a value when it should be switched off. That way, we'd have some hysteresis and the output should work reliably. For example: You want your drybox at around 35% humidity, so you start your heater at humidity 38%, and switch it of when it reaches 25% (keep in mind that the AM2302 measures relative humidity, so the value will drop dignificantly when temperature rises. All values are rough estimates, I have not tested this scenario).

sajovicd, 2020-12-04:
Your example is already implented for a long time and it works perfect I use all the time for my dehumidifier box

TazerReloaded, 2020-12-04:
Oh, didn't see that I guess, I don't have a dehumidifier in my box. Thought only temperature could be used as an input. Well, then only the offset problem remains. Although it's not essential, because even readings with wrong offset can be used for controlling outputs.

Mikelomi, 2020-12-04:
Hi,first thanks for your quick response. At the moment it's not possible to use the Humidity to Control any switch by your Plug in as I see. I only think it would be a nice Feature for the Future. Could be a misunderstanding about my English. Thank you. M.