vitormhenrique/OctoPrint-Enclosure #381: Output shell script

BoergiM87, 2020-12-28:
**Describe the bug**
Hi, if I create an output "shell script" with the path to the script and execute it, the whole webinterface does not react anymore and I have to restart the PI.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a output "shell script" with the path of it
2. Click on the execute bottom of the output
3. See error

**Additional context**
[image]

SaKiEQ, 2021-01-05:
I am doing pretty much the same, except with python scripts and not having any issues... 

Does you shell script execute when called directly from a terminal ? It also depends what your script contains... 
[image]

BoergiM87, 2021-01-05:
I have test it to with a python script, but the same problem. Octoprint hangs completely and you have to restart the Pi.

Over the terminal i can execute the script with no errors.

This is the script:
`
import requests
url = 'XY'
headers = {
    'Accept': 'application/json, text/javascript, */*; q=0.01',
    'X-Requested-With': 'XMLHttpRequest',
    'Content-Type': 'application/json; charset=UTF-8'
}
payload = ''
r = requests.post(url, data=payload, headers=headers)
print r

`

luizbgomide, 2022-02-28:
I was having this issue with octoprint hanging. Now I recreated everything and it doesn't hang, but also doesn't work... :/

I create a script with a just an `echo 'test'` and it doesn't output anything....

[image]

vitormhenrique, 2022-02-28:
Shell script is only supposed to run the script, not to output anything that it spits out.
https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/master/octoprint_enclosure/__init__.py#L534

vitormhenrique, 2022-02-28:
@BoergiM87 that is a PYTHON script, and this is not supposed to run a PYTHON script, only a SHELL script.

luizbgomide, 2022-02-28:
@vitormhenrique my issue was that it doesn't seem like the script is actually running.
I the end I removed all my settings and recreated the call to the scripts and it worked... I'm really not sure of why it didn't work before.

I'm using that to trigger PSU Control, only because it has a dedicated icon on Printoid, otherwise it could have managed it entirely in your plugin...

DiP-ER, 2023-01-18:
> @vitormhenrique my issue was that it doesn't seem like the script was actually running. In the end I removed all my settings and recreated the call to the scripts and it worked... I'm really not sure of why it didn't work before.
> 
> I'm using that to trigger PSU Control, only because it has a dedicated icon on Printoid, otherwise it could have managed it entirely in your plugin...

@luizbgomide Did you ever figure out what the problem was besides removing your settings and re-creating them?  I'm having the same issue trying to call:

curl -s -H "Content-Type: application/json" -H "X-Api-Key: YOUR_API_KEY" -X POST -d '{ "command":"turnPSUOn" }' http://YOUR_OCTOPRINT_SERVER/api/plugin/psucontrol

I created a script to touch a file which works, but doesn't like the above line.

luizbgomide, 2023-01-18:
@DiP-ER no, no idea.