vitormhenrique/OctoPrint-Enclosure #383: Filament Sensor to Pause Print Results in NO ACTION.

Cortexian, 2021-01-05:
I have a Creality Ender 5 Plus. Using the latest Kersey Fabrications firmware, basically the stock Creality firmware but with a fixed probe offset...

**To Reproduce**
Steps to reproduce the behavior:
1. Start Print.
2. Run out of filament.
3. Filament sensor falls (verified detected using your troubleshooting python script).
4. Print continues.
5. Print never pauses.

**Expected behavior**
Print pauses, as configured, when filament sensor falls. Then runs the "After print is paused" script in the Octoprint settings.

**Additional context**
Here's some configuration info:
[trimmed-octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/5767930/trimmed-octoprint.log)

"After print job is cancelled" script (from Octoprint examples):
>{% if pause_position.x is not none %}
>; relative XYZE
>G91
>M83
>
>; retract filament, move Z slightly upwards
>G1 Z+5 E-5 F4500
>
>; absolute XYZE
>M82
>G90
>
>; move to a safe rest position, adjust as necessary
>G1 X175 Y350
>{% endif %}

"Before print job is resumed" script (also from Octoprint documentation examples):
>{% if pause_position.x is not none %}
>; relative extruder
>M83
>
>; prime nozzle
>G1 E-5 F4500
>G1 E5 F4500
>G1 E5 F4500
>
>; absolute E
>M82
>
>; absolute XYZ
>G90
>
>; reset E
>G92 E{{ pause_position.e }}
>
>; move back to pause position XYZ
>G1 X{{ pause_position.x }} Y{{ pause_position.y }} Z{{ pause_position.z }} F4500
>
>; reset to feed rate before pause if available
>{% if pause_position.f is not none %}G1 F{{ pause_position.f }}{% endif %}
>{% endif %}

[image]
[image]
[image]

I'm using the stock Ender 5 Plus filament sensor. The weird thing is that when I first set this up (without GCODE scripts), the print paused perfectly (just froze in place, which I didn't like, hence the scripts).

It's 100% detecting the filament runout every time I remove it from the sensor manually while running the troubleshooting python script:
[image]

Cortexian, 2021-01-06:
Update: The sensor is definitely working. I redid all my wiring connections today with correct DuPont connectors to attach to the GPIO pins. I'm getting emails from IFTTT when I do filament changes.

Pause just isn't triggering for some reason.

vitormhenrique, 2021-01-11:
I'm working on a new verison of the plugin with python 3 compatibility and better logging etc, we will need that in order to get some logging and see if the plugin is sending commands to the printer. If it is and the printer is not responding there is no much I can do. 
Because of the major refactor for python 3 I'm not going to support this legacy python 2 plugin anymore, and the new version will include better configuration interface and embedded libraries for temperature sensors etc...
Unfortunately it's a balance of how much I can code a day with my day job, masters and other projects, so for now I would recommend either you trying to add some logs on the source code and debug or wait for a new release in a few months.

mockendon, 2021-01-13:
Thank you for your hard work on this - a key part of the Octoprint functionality.  UnfortunatelyI have the same issue.  I have set up scripts in octoprint that handle pause and resume very well.  I would like my filament sensor to trigger an octoprint 'pause' just like pressing the pause button and my options are set exactly as above.  Like Cortexian nothing happens when the sensor triggers in this configuration.