vitormhenrique/OctoPrint-Enclosure #389: DS18B20  [enforced open drain please flag it properly in dt/acpi dsdt/board file] OctoPrint-Enclosure

firozdang, 2021-01-31:
Which Plugin? 
https://github.com/vitormhenrique/OctoPrint-Enclosure

What's wrong with the plugin listing?
I connected DS18B20 sensor with pullup resisters, and after updating /boot/config with dtoverlay=w1-gpio, I cant see anything in dmesg
when I do dmesg | grep gpio, I got the following error:
enforced open drain please flag it properly in dt/acpi dsdt/board file

Using the following sensor kit, I checked with multimeter, vcc and gnd are correctly connected, also my data signal is correctly connected on GPIO#4, PIN#16:

https://www.banggood.com/DS18B20-Temperature-Sensor-Module-Kit-Waterproof-Electronic-Building-Block-p-1027637.html

ademuri, 2021-10-07:
I think that error is a red herring - my sensor works fine and I also see that error. Are you sure you have it connected to the right pin? On my Pi 3B+, GPIO 4 is pin 7.

luizbgomide, 2022-01-30:
I also have that warning, not sure what to do about it or how to fix it? I think we are probably missing something in the config.txt, but looting at the dtoverly docs I don't see what could be missing...

BTW: I tried to two dtoverlays: gpio-w1 and gpio-w1-pullup. No change, I thought about removing the external resistor pullup. Maybe that is unnecessary now?

tyukesz, 2022-11-28:
I have the same log

DavyGD, 2024-08-01:
I have the same.