vitormhenrique/OctoPrint-Enclosure #409: Si7021 no reading

Susterki, 2021-03-05:
Sorry I'm am new at this. 

I connected the Si7021 sensor (VIN-pin1 GND-pin6 SCL-pin5 SDA-pin3).
I did this:

Run sudo raspi-config
Use the down arrow to select 9 Advanced Options
Arrow down to A7 I2C
Select yes when it asks you to enable I2C
Also select yes when it asks about automatically loading the kernel module
Use the right arrow to select the button
Select yes when it asks to reboot
Install some packages:

sudo apt-get install i2c-tools python-pip
Find the address of the sensor:

i2cdetect -y 1

[image]

I went in Octoprint and did this

[image]

and I get this

[image]

I toggeld "use sudo" in "advanced options" nothing changes

here is the log, I think
[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/6090908/octoprint.log)

what am I doing wrong?

FritzWlan, 2021-04-18:
I have the exact same issue. Also tried with sudo on and off (advanced settings). Did not change anything. Removing sudo password did not help either. 

Log says it expects two values to be generated but could only find one (ValueError).

ab-tools, 2021-04-26:
Same here as well, any further ideas on this?

andiatbaden, 2021-05-23:
Hello,

got the same issue. Additional i got no seperate tab for the plugin
[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/6528399/octoprint.log)

thanks for your support

bigmac5753, 2021-06-21:
has anyone found a fix for this?