vitormhenrique/OctoPrint-Enclosure #414: BME280 doesn't work if placed in another i2c Bus different than 1

lpla, 2021-03-18:
**Describe the bug**
I installed a BME280 board through GPIO pins 23 and 24, because i2c standard pins are used by Fan Shim, in my Raspberry Pi 4 using OctoPi. I configured the sensor in the OctoPrint plugin by specifying the i2c Bus as 11, which is the one being detected by `i2cdetect -y 11`. I tried running in terminal `~/oprint/lib/python3.7/site-packages/octoprint_enclosure $ python3 BME280.py 76` and it didn't work until I edited line 14 replacing `bus = smbus.SMBus(1)` with `bus = smbus.SMBus(11)`. But even that didn't make it work through OctoPrint, showing 0ºC and 0%.

**To Reproduce**
Move a working BME280 from standard i2c bus 1 to another one set up in `/boot/config.txt`. Test it with the commands written above and other example scripts from tutorials to check that the BME280 works. Check that OctoPrint is unable to show the module values.

**Expected behavior**
It should work after defining i2c Bus as 11 in Enclosure Plugin 'Add Input' setup.

**Additional context**
[image]

[image]

[image]

ddB0515, 2021-05-13:
I had similar issue after update (try go to logs and check what is is saying)

I had to install smbus/smbus2 (bot user and sudo)
`pip3 install smbus`
`pip3 install smbus2`

restarted and worked like charm, try and post result or at least log :)

MircoMaker, 2021-09-12:
Same here, used 2x BME280 and tried first with 2x 1c2 bus and second with TCA9548A. Both times just first bus is used and temperature of first bus is shown for both.

EFXPROMedia, 2023-08-21:
I am having the same issue, my sensor is on `0x76` I tried installing the 2 smbus and still dont have a temperature showing on the enclosure plugin UI, the sensor works because I can get its status, see below
```bash
pi@octopi:~ $ python bme280.py
Chip ID     : 88
Version     : 1
Temperature :  19.04 C
Pressure :  1009.35522279 hPa
Humidity :  0.0 %
```

morphias2004, 2025-02-17:
Thank-you @lpla and @ddB0515  !!!!!    👍 

The combo of installing smbus2, editing the BME280.py file and tuning off Use SUDO fixed it.