vitormhenrique/OctoPrint-Enclosure #416: temp_hum_control cooler Auto Startup (when print finishes)

wickedbeernut, 2021-03-22:
The Enclosure plugin 'temp_hum_control' controllers currently support several practical startup and shutdown options,

- _Start with server - Choose if output should turn on automatically when octoprint starts_
- _Auto Startup (when print starts) - Choose if output should turn on automatically when print starts_
- _Auto Shutdown - Choose if output should turn off automatically when print finishes_

Please add an option to automatically turn **on** a 'cooler' controller output when the print **finishes**,

- _Auto Startup (when print finishes) - Choose if output should turn on automatically when print finishes_

The controller would cool until the target temperature (temp_ctr_set_value) **or** a maximum (configured) duration is reached.

This would be used in conjunction with a heatbed cooling fan or an air purifier (e.g., HEPA / activated-carbon filter) to purge the air from an enclosure.

Refer to the "temp_hum_control cooler heatbed Temperature Sensor" feature request.

Dak0r, 2021-03-22:
I'd like to see something like ```set [on / off] on [Printer Event Type] ```
with[] being options to choose from.

Printer Event Types could be "Print Started", "Print Finished", "On Error".
This would allow to trigger an enclosure fan on print finish (for which I'm currently using a different plugin) but also an Automatic Power-Off on Error or unexpected disconnect, which would be a huge safety improvement IMO.

wickedbeernut, 2021-03-23:
I'm certainly not opposed to satisfying this feature request in the context of a more general framework.  However, I'm looking for more than just toggling an output on and off.  I'm looking to enable a cooler controller until a target temperature is reached or a maximum period of time has elapsed at which time the controller would be disabled.  I think this could be generalized to heater and dehumidifier controllers.  For example, how about preheating the enclosure and/or filament dryers to a target temperature before a print starts?  This could be accomplished using your proposed framework, we'd just need "set" values other than Output On and Output Off (e.g., Control to Target, Control for Time, Control to Target with Timeout, ...).

By the way @Dak0r, on a related subject, did you read my suggestion to add target temperatures and humidities to your `octoprint.comm.protocol.temperatures.received` hook?  This is in support of Enclosure plugin heater, cooler and dehumidifier controllers.