vitormhenrique/OctoPrint-Enclosure #423: NavBar-Button for shell script not available

eloo, 2021-06-06:
**Describe the bug**
Hi,
i have configured a shell script as output and activated the "Show Button on Navbar" option.
But i cant find the button in the navbar. 

**To Reproduce**
Steps to reproduce the behavior:
Create a shell output like this:
[image]

**Expected behavior**
I would expect some kind of a button in the navbar. But i can only see the button in the enclosure plugin tab.

**Actual behavior**
[image]

**Additional context**
OctoPrint 1.6.1 
Python 3.7.3 
OctoPi 0.18.0
Enclosure Plugin 4.13.1

michaelbeljaars, 2022-12-21:
Yeah, I have the same thing. I am using two shell scripts to turn on and off a LED strip. I tried enabling the add to NavBar functionality for easier access, but nothing happens.

zach-ovic, 2023-03-18:
Same here. But it looks like nobody is maintaining this anymore, right? :(

michaelbeljaars, 2023-03-19:
> Same here. But it looks like nobody is maintaining this anymore, right? :(

Looks like it indeed. It's too bad, I haven't found an other plugin that has the same functionality.