vitormhenrique/OctoPrint-Enclosure #424: DHT22 unable to view readings

davbcr, 2021-06-14:
Despite proper configuration of the plugin it's not able to show any readings from DHT22 properly connected and working.

[image]
[image]

SUDO in advanced settings is disabled
getDHTTemp.py is below:

import sys
import Adafruit_DHT

# Parse command line parameters.
sensor_args = { '11': Adafruit_DHT.DHT11,
    '22': Adafruit_DHT.DHT22,
        '2302': Adafruit_DHT.AM2302 }
if len(sys.argv) == 3 and sys.argv[1] in sensor_args:
    sensor = sensor_args[sys.argv[1]]
    pin = sys.argv[2]
else:
    sys.exit(1)

humidity, temperature = Adafruit_DHT.read_retry(sensor, pin,2,0.5)

if humidity is not None and temperature is not None:
    print('{0:0.1f} | {1:0.1f}'.format(temperature, humidity))
else:
    print('-1 | -1')

sys.exit(1)

Can you help me modify getDHTTemp.py to get readings? i've noticed that with the script of Raspcontroller App here attached it works.
[dht_v6.txt](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/6646952/dht_v6.txt)

noz1380, 2021-06-15:
I have the same issue, and wanting to fix.  I'm not sure if this is still supported?

I have read that there is a change in an update, and adafruit dont support the old DHT, but the newer Adafruit_CircuitPython_DHT

noz1380, 2021-06-15:
Please take a look at this thread.

https://stackoverflow.com/questions/63232072/cannot-import-name-beaglebone-black-driver-from-adafruit-dht

I added this line as per the instructions

elif match.group(1) == 'BCM2711':
    return 3

It works!!

davbcr, 2021-06-15:
> Please take a look at this thread.
> 
> https://stackoverflow.com/questions/63232072/cannot-import-name-beaglebone-black-driver-from-adafruit-dht
> 
> I added this line as per the instructions
> 
> elif match.group(1) == 'BCM2711':
> return 3
> 
> It works!!------Sorry it works on command line, which is better than I had before.
> 
> But not in Octoprint.

Mee too. it's not working via Octoprint.

noz1380, 2021-06-15:
What I cant figure out is that the example code to show temp and humidity (sudo ./AdafruitDHT.py 2302 4) in terminal is pretty much identical to the code in "getDHTTemp.py" which leads me to think there is a problem elsewhere.

unsplorer, 2021-07-10:
I'm running OctoPrint in a docker container and was able to resolve this issue by installing the newer Adafruit_CircuitPython_DHT library and python3 inside the container. I had to adjust the call to getDHTTemp.py inside octoprint_enclosures __init__.py to use python3. I then had to modifiy getDHTTemp.py for the newer CircuitPython library as follows :
`import sys
import adafruit_dht
from board import D19 #my sensor is connected to D19 on RPi. adjust your pin accordingly
import time
# Parse command line parameters.
sensor_args = { '11': adafruit_dht.DHT11,
    '22': adafruit_dht.DHT22}
#        '2302': Adafruit_DHT.AM2302 } depreciated
if len(sys.argv) == 3 and sys.argv[1] in sensor_args:
    sensor = sensor_args[sys.argv[1]]
    pin = sys.argv[2] #ignoring this for testing purposes
else:
    sys.exit(1)

dht_device = adafruit_dht.DHT22(D19)
read_attempts = 0
while read_attempts < 7: # try for a reading 7 times since dht's are unreliable
        try:
                humidity = dht_device.humidity
                temperature = dht_device.temperature

                if humidity is not None and temperature is not None:
                    print('{0:0.1f} | {1:0.1f}'.format(temperature, humidity))
                    sys.exit(1)
        except RuntimeError as error:
                read_attempts = read_attempts + 1
                time.sleep(2.0)
                continue
        except Exception as error:
                read_attempts = read_attempts + 1
                dht_device.exit()
                raise error

        time.sleep(2.0)

print('-1 | -1')
sys.exit(1)`

unsplorer, 2021-07-10:
I'm running OctoPrint in a docker container and was able to resolve this issue by installing the newer Adafruit_CircuitPython_DHT library and python3 inside the container. I had to adjust the call to getDHTTemp.py inside octoprint_enclosures __init__.py to use python3. I then had to modifiy getDHTTemp.py for the newer CircuitPython library as follows :
  ```import sys
   import adafruit_dht
   from board import D19 #my sensor is connected to D19 on RPi. adjust your pin accordingly
   import time

   # Parse command line parameters.
   sensor_args = { '11': adafruit_dht.DHT11,
       '22': adafruit_dht.DHT22}
   #        '2302': Adafruit_DHT.AM2302 } depreciated
   if len(sys.argv) == 3 and sys.argv[1] in sensor_args:
       sensor = sensor_args[sys.argv[1]]
       pin = sys.argv[2] #ignoring this for testing purposes
   else:
       sys.exit(1)
  
   dht_device = adafruit_dht.DHT22(D19)

   read_attempts = 0
   while read_attempts < 7: # try for a reading 7 times since dht's are unreliable
          try:
                  humidity = dht_device.humidity
                  temperature = dht_device.temperature
  
                  if humidity is not None and temperature is not None:
                      print('{0:0.1f} | {1:0.1f}'.format(temperature, humidity))
                      sys.exit(1)
          except RuntimeError as error:
                  read_attempts += 1
                  time.sleep(2.0)
                  continue
          except Exception as error:
                  read_attempts += 1
                  dht_device.exit()
                  raise error
  
          time.sleep(2.0)
  
   print('-1 | -1')
   sys.exit(1)