vitormhenrique/OctoPrint-Enclosure #435: The future of OctoPrint-Enclosure

hbast, 2021-08-30:
Hey @vitormhenrique, I'm currently building a case for my 3D printer and I found this plugin while putting together the software. I am confused about the status of the plugin. I don't see any real progress in development since some time. In the Octoprint plugin manager, version 4.13.1 is current; however, this does not appear to exist in this repository - or I'm blind ;). There is only 4.12. In addition, some pull requests are open with new features, such as connection of new sensors, or graphical functionalities, which are not processed / incorporated. I need the integration of BME688, which is interface compatible wiht BME680... I could help testing and integrating the PR.

For me, the question is whether it is worth using this plugin, or to realize the control of my 3D printer enclosure in another way. Where are there problems? How can the community contribute to revive and develop the plugin. What does the roadmap look like?

stephenfeather, 2021-09-06:
Same here.  With anything other than PLA, need to test AQ inside enclosures.
Also adding a laser head which gives off contaminants as it burns.

@hbast maybe a fork is in order if you don't get an answer soon (its been a week)

mynameisbrendan, 2021-10-13:
First up - I am not a developer, so I offer no help here.  Sorry!

I just happened to get the alert above in my octoprint UI at the same time as I was offered the OP update to 1.7.0.  I just wanted to check if that is a coincidence, or if this thread has come about due to an incompatibility of the current version of Enclosure Plugin with OP 1.7.0?

jneilliii, 2021-10-13:
There is currently an issue with the current release of the Enclosure plugin and OctoPrint 1.7.0. I've submitted a PR that I believe will resolve the issue, but awaiting verification from #432.

ni-aackerman, 2021-10-13:
Maybe @wmonzel wants to help?

wmonzel, 2021-10-13:
@foosel @vitormhenrique @ni-aackerman Sure, count me in!  I'm happy to help in any way I can.

MIIB1, 2021-10-21:
Hello,

i want ask can i control snapmaker 2 A350 enclosures ?

i can't find how do it ?

nelsonov, 2021-11-04:
Somebody should pin this issue.  The link from the Octoprint notification landed me here, but that may not be the case for everyone.

Also, the invite above for the Discord channel is invalid.  Is the channel no longer being used or did it just expire?  I have so many future plans for this plugin that I hope I can find a way to contribute back.

AlexDep, 2021-11-04:
hi,
Here a permanent link : https://discord.gg/bMQCsScaus

DIY89, 2022-01-07:
Hi folks, 
first of all. I really like this plugin it. So thanks to all you greats guys who develop this.
During my work on my 3D Printer i recognized that the lower level IO Stuff is working much better on an Arduino Board like 1-Wire communication and much more safely for controlling relays.
So i would like to contribute this great project but i need a little start help to initialize the git repository. I there a chance to run the repo directly from GitHub?

Thanks

vitormhenrique, 2022-01-07:
@DIY89 I have no idea what do you mean by run the repo directly from github. 

you mean install from github? run from a local copy on octoprint?

DIY89, 2022-01-07:
@vitormhenrique: Currently i've installed the Plugin via `oprint/bin/pip3 pip install "https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/master.zip`. So i find the files i want to change under `oprint/lib/python3.7/site-packages/octoprint_enclosure`. For example the `__init__.py`. How can i bring the changes back to your `master branch` via git?

rtheil-growlink, 2022-01-20:
> @vitormhenrique: Currently i've installed the Plugin via `oprint/bin/pip3 pip install "https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/master.zip`. So i find the files i want to change [...]

1. clone repo
2. create new local branch
3. make changes
4. when happy, commit changes and push to github
5. create a pull request for the code in the branch
6. @vitormhenrique will approve & merge your changes if he so chooses

NDR008, 2022-02-15:
I was using OctoRelay, and after fixing fundamental bugs in that plugin (which never good pulled into the main master), I came across your plugin.
Seems like a WAY better baseline.
So I think I'll contribute to yours instead.

First thing I wish to add is, an autoshutdown timer (to turn things off with delays after printing). :)

NDR008, 2022-02-16:
> @vitormhenrique @DIY89 
Read up on git. Fork this repo, make modifications, push them to your fork, then make a pull request to this master repo.

ni-aackerman, 2022-02-17:
> I was using OctoRelay, and after fixing fundamental bugs in that plugin (which never good pulled into the main master), I came across your plugin. Seems like a WAY better baseline. So I think I'll [...]

The timers with delay would be extremely helpful, since some things are activated with the inverted state, so there is no way of turning them off after the print ends, only start them.

vitormhenrique, 2022-02-17:
There is already a timer based shutdown system implemented.

JeremyLaurenson, 2022-02-17:
Hey @vitormhenrique 

There is some discussion in the Discord chat server around pull requests, for example the one 'ready to go' in November and whether we should just fork a new build.

What are your current thoughts?

ni-aackerman, 2022-02-17:
> 

How would you configure then in this particular case to stop after 5 minutes. This is an Active Low setup, so the fan turns off (on) when the print ends? For me it doesnt work when I add the time in seconds.
[image]
.

troywneal, 2022-02-19:
Hi there. While I am not a programmer by trade, I spent several years programming in python for work. While it has been a while since, I still regularly program in other languages and I am very interested in helping (and learning). I tried to join the discord, but it was closed. I may be late to this, but I hope to contribute to your amazing work.

hbast, 2022-02-22:
After all the good will and great support from @foosel  via Twitter, a lot of people showed up on the Discord server. All of them were ready to support in their own way. Be it in project management, Python programming, organization, hardware related programming, expertise was gathered there from all corners. But it just didn't happen. People were stalled and nothing was done instead of assigning tasks and facilitating them.

I for myself am fed up with it. It's a disgrace. This horse is dead and should not be resurrected.  Let it rest and saddle up on another one.

closed 2022-02-22 by hbast

troywneal, 2022-02-22:
> <vitormhenrique/OctoPrint-Enclosure/issue/435/issue_event/6119151554@
> github.com>

I understand that. Thank you for taking the time to respond. Also, thank
you for all of the time you put in. I am very grateful as a user. I am also
very sorry that we, as a community, weren't able to come through on this.

Again, thank you for your time.

mhesselgator, 2022-02-25:
I'm bummed!  Cool plugin!!!

spiff72, 2022-02-25:
I am a bit confused - I got signed up for notifications on this topic a while back, and I am not clear on whether this plugin is still viable.  From reading this thread, it seems it isn't.  However, it is working fine for me right now.
I have Octoprint 1.7.3
Enclosure Plugin 4.13.2
I have it set up to turn on/off an exhaust fan at certain temperatures based on some starting GCODE that gets added when i slice parts (based on the tool temp (PLA material sets a lower enclosure temp than stuff like PETG/ABS/PCblend).  Everything works great on my end.

rtheil-growlink, 2022-02-25:
> I am not clear on whether this plugin is still viable

It still works really well, but there are a lot of issues noted that are incompatible with Python 3, and there have been no updates. Even pull requests are not being merged.

spiff72, 2022-02-25:
> > I am not clear on whether this plugin is still viable
> 
> It still works really well, but there are a lot of issues noted that are incompatible with Python 3, and there have been no updates. Even pull requests are not being merged.

Ah - OK.  I had updated octoprint a while back and since it was using Python 3 now (and I was still using the Enclosure Plugin successfully) I thought all had been resolved.

Thanks!

GerhardMassenbichler, 2022-02-25:
The Future - what‘s needed:
- design principles (API, PigPIO vs. small libs, etc.)
- CI/CD for automated builds  
- automated tests to secure the codebase 
- People with write access to merge pull-requests
- short releasecycle 
- create a EnclosureNG-Fork? 

Anything else?

vitormhenrique, 2022-02-26:
i think there is a misunderstanding here.

I said that I was gonna make this plugin compatible with python 3, and we got those pull requests on.

I'll accept PR's for but fixes but NOT new features.

The current way that the plugin works takes a lot of effort and knowledge of python environments and libraries to make it work, the configuration is terrible and user's just don't know how to use it.

The idea here is to move away from this plugin and build another with helper plugins, each helper plugin would have all libraries and easier way to configure them.
For example a GPIO helper plugin would be able to act like a trigger and actions.
The main plugin would have the glue logic and enable a screen like an "IF THIS THEN THAT".

The POC with the new style of plugin was here on the "Python 3" branch. But moved it to here: 
https://github.com/vitormhenrique/OctoPrint-Enclosure-V2

An example of a helper plugin is here:
https://github.com/vitormhenrique/OctoPrint-Enclosure-RPiGPIO-Faker

This is only a POC and I already have some stuff that I don't like and will be changing. 
Please this is a free software that I worked for years, don't come saying "it's a shame" or whatever, specially if you have zero contributions to this plugin or any opensource library.

TyriqueDaniel, 2022-04-15:
Hello sorry here really looking forward to helping I'm sorry don't have much time couldn't read the entire thread but the discord link is invalid. Please let me know if you are still looking for help thank you

jneilliii, 2022-04-15:
Updated invite: https://discord.gg/cwgqHnEZk3

spiff72, 2022-05-20:
I know this is closed, but is it reasonable to say "Don't upgrade to octoprint 1.8.0 if you are still using OctoPrint-Enclosure"?

stephenfeather, 2022-09-04:
When large frameworks or underlying languages have major version changes, it often requires a lot of code changes.

I'm currently fighting a PHP 8x, MySQL 8x battle. Neither side paid attention and the 2 giant ships passed each other.

Python 2 -> Python 3 was an 8 year span with nearly 14 years to present since 3 was released.
But here is the thing to remember on all sides here.  Until Octoprint moved to Python 3, plugins couldn't.

Until just a few years ago, Apple was shipping Python 2 in OSX and it was (and for some still is) a PITA to get python 3.x on.

@vitormhenrique has put a ton of time into a plugin that by all intents was designed to take a huge collection of disparate inputs (hardware, software, cloud) and bring them into a view that we could read easily.  And for that I think him tremendously.

On the other side as a user, I understand the frustration when a working setup just dies on you because of underlying changes.  You want the latest Octoprint, the fixes, the advantages, and now a sub-product that depends on that breaks from the changes.

Python is definitely not in my wheelhouse.  "Hack in" fixes I can do sometimes. Give me some PASCAL and I'm a KING! :)

I read through the POC and like the direction.

Now, I want the plugin to grab an alexa hook and sound an alarm throughout the property that a 3d printer is on fire! j/k

> i think there is a misunderstanding here.
> 
> I said that I was gonna make this plugin compatible with python 3, and we got those pull requests on.

> This is only a POC and I already have some stuff that I don't like and will be changing. Please this is a free software that I worked for years, don't come saying "it's a shame" or whatever, specially if you have zero contributions to this plugin or any opensource library.

nicman23, 2022-12-25:
fixed dht sensors for me on python 3

```
import sys
import time
import board
import adafruit_dht

# Parse command line parameters.
sensor_args =   {
                    '11': adafruit_dht.DHT11,
                    '22': adafruit_dht.DHT22,
                    '2302': adafruit_dht.DHT22
                }

if len(sys.argv) == 3 and sys.argv[1] in sensor_args:
    sensor = sensor_args[sys.argv[1]]
    pin = "D%s" % sys.argv[2]
    pin = getattr(board,pin)
else:
    sys.exit(1)

dhtDevice = sensor(pin)

# DHT sensor read fails quite often, causing enclosure plugin to report value of 0.
# If this happens, retry as suggested in the adafruit_dht docs.
max_retries = 3
retry_count = 0
while retry_count <= max_retries:
#    try:
     if True:
        humidity=dhtDevice.humidity
        temperature=dhtDevice.temperature

        if humidity is not None and temperature is not None:
            print(('{0:0.1f} | {1:0.1f}'.format(temperature, humidity)))
            sys.exit(1)
#    except Exception as e:
#        print('-1 | -1')
#        sys.exit(1)

     time.sleep(1)
     retry_count += 1

print('-1 | -1')
sys.exit(1)
```

nrafonso, 2023-02-21:
Hi everyone, I'm struggling to get the DHT22 sensor working on the Enclosure plugin. Followed the documentation on github for this but seems not working. I've run a small program on the PI terminal and the sensor work fine there. I have the raspberry 4b with OctoPrint version 1.8.6 and OctoPi version 0.18.0. Thank you for your help

nrafonso, 2023-02-25:
Thanks @nicman23 
I have updated the `/home/$myuser/oprint/lib/python3.9/site-packages/octoprint_enclosure/getDHTTemp.py` with your code and it solved things.

FYI on the plugin page I have sudo disabled because I'm already able to get temperatures with normal user.

mhesselgator, 2025-03-27:
I've never found an alternative for this plugin; it is incredible with the ease and fore thought given to the user.  I'd love to see someone revive the development of this plugin!  I'd love to help but I have limited programing capabilities.

vitormhenrique, 2025-03-27:
@uipko feel free to fork and bring the plugin back to life. 
At a point I started working on V2 with much better architecture using plugins, but to be honest I lost interest. Most new 3d printers have either enclosures / lights already and I'm just not excited to work on this project again. 
I had minimal incentive to work on this and was dealing with supporting a lot of users at a point in time. I thought about doing another version of the plugin with more features and maybe commercializing it somehow to have a financial incentive to work on it, but as newer printers are going more and more to the closed ecosystem I don't think that this will have enough traction. 
The plugin as it is, is defunct, and I don't plan to resurrect it.

uipko, 2025-03-30:
@vitormhenrique Thanks for the quick response! I guess you have a fair point in that a lot of new/current printers do come with an enclosure. I'm still using an old printer with homemade enclosure.

I'll create a fork and try to revive the parts that I need. Thanks a lot for the great work, it is a pretty extensive code base!

nfored, 2025-03-31:
I also will say this plugin had features that factory enclosures don't have, I don't have a lot of factory enclosed printers a T1 Pro, and a K2 Plus.  

However the custom enclosure was more feature rich, dedicated door locks to prevent children from opening the printer and harming themselves, the ability to use Pan/Tilt camera that can track the print rather than show a fixed shot.  The ability to only turn on lights when it prints turn them off when its not printing, or use a light that act as a flash for octolapse so the lights are off always except for a second prior to an image being taken. 

This was literally the most useful tool because it allowed such an amazing level of flexibility.