vitormhenrique/OctoPrint-Enclosure #446: temperature / humidity control does not auto shutdown heater after print

pintail120, 2021-10-29:
I have X2 DS18B20's and X1 Si7021 on my octopi
A clear and concise description of what the bug is:
I use two different heaters in my cabinet, one is powered by mains(fan) for quick heating
The second heater is static with no fan to minimise disruption on ABS printing.
I have configed both heaters to "auto shutdown", however only the mains heater clears its set temp and turns off.
The static heater continues to operate even though it has been set to "auto shutdown"
Both heaters are configured to track the Si7021 as the temperature source for actions.

**Expected behavior**
Both heaters should turn off, but the one does not.
**Additional context**
[image]
se.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299) here.

pintail120, 2021-10-31:
it just did this again, logs uploaded.

[octoprint-systeminfo-20211031193728.zip](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/7449649/octoprint-systeminfo-20211031193728.zip)

pintail120, 2021-11-01:
this morning a print completed, however the 12V heater did follow auto shutdown, remained stuck on
[octoprint-systeminfo-20211101065900.zip](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/7451241/octoprint-systeminfo-20211101065900.zip)
Any ideas why one of the enclosure plugins remain on even though the Autoshutdown is activated ?

vitormhenrique, 2021-11-04:
I'm completely re-writing this plugin, and will stop supporting the old one soon. 

So unfortunately will not be looking to fix issues on this. I have the same setup on my rpi with no issues, so i can't put a lot of time looking for this and getting the new plugin ready.

closed 2021-11-04 by vitormhenrique

pintail120, 2021-11-04:
sure no problem
does that mean the conflict with touch ui and the charting plugin will get
resolved ?
what version of python are you writing this in ?