vitormhenrique/OctoPrint-Enclosure #448: DHT11 driver not matching Adafruit_DHT

MiataMike, 2021-10-31:
Hi All, I'd like to help contribute, but I'm not sure exactly how. I fixed the script to match some different names. Should I submit a pull request?

DHT11 would work using the example code from adafruit, but not in the plugin

Followed the cloning instructions to install the DHT library, but the getDHTTemp.py file didn't match the library (adafruit_dht vs Adafruit_DHT)

Updating the library name and function has the sensor showing up in the nav bar as expected

debug log
2021-10-31 08:53:53,035 - octoprint.plugins.enclosure - DEBUG - Sensor Temp Temperature: 0.0 humidity 0.0
2021-10-31 08:54:03,046 - octoprint.plugins.enclosure - DEBUG - Temperature dht cmd: python3 /home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/getDHTTemp.py 11 4
2021-10-31 08:54:03,274 - octoprint.plugins.enclosure - DEBUG - Dht result: b''
2021-10-31 08:54:03,274 - octoprint.plugins.enclosure - INFO - Failed to execute python scripts, try disabling use SUDO on advanced section of the plugin.
2021-10-31 08:54:03,275 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('not enough values to unpack (expected 2, got 1)',)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 1113, in read_dht_temp
    temp, hum = stdout.decode("utf-8").split("|")
ValueError: not enough values to unpack (expected 2, got 1)
2021-10-31 08:54:03,275 - octoprint.plugins.enclosure - DEBUG - Sensor Temp Temperature: 0.0 humidity 0.0

`import sys
import Adafruit_DHT

# Parse command line parameters.
sensor_args =   {
                    '11': Adafruit_DHT.DHT11,
                    '22': Adafruit_DHT.DHT22,
                    '2302': Adafruit_DHT.DHT22
                }

if len(sys.argv) == 3 and sys.argv[1] in sensor_args:
    sensor = sensor_args[sys.argv[1]]
    pin = sys.argv[2]
else:
    sys.exit(1)

#dht_dev = sensor(pin)
#humidity = dht_dev.humidity
#temperature = dht_dev.temperature
humidity, temperature = Adafruit_DHT.read_retry(sensor, pin)

if humidity is not None and temperature is not None:
    print('{0:0.1f} | {1:0.1f}'.format(temperature, humidity))
else:
    print('-1 | -1')

sys.exit(1)
`

Albertowue, 2021-11-02:
The current getDHTTemp.py file works with the new   Adafruit_CircuitPython_DHT library, your changes are needed to make the script work with the old Adafruit_Python_DHT library. I think the best thing is to install the new library and not modify the getDHTTemp.py

vitormhenrique, 2021-11-04:
I merged the #450 so will close this, as the new library fixes the issue.

closed 2021-11-04 by vitormhenrique