vitormhenrique/OctoPrint-Enclosure #449: Not sure if this would be a bug report or feature request........

poppabear1000, 2021-10-31:
I noticed that on "Outputs" if I have lets say a temp alarm that turns on an output above some temperature, that once it turns on, it does not turn off once it drops below that temperature again.

I.E.  I have an exhaust fan, that if my chamber is set to 70C, but for some reason the heater overshoots that temp, and my temp alarm is set to 72C, so the then the exhaust fan turns on, to help bring the temp in the chamber back to 70C, but once the temp drops below 72C the fan (temp alarm) does not turn off, so then the heater has to "Fight" the exhaust fan.  
I have another temp alarm (similar situation) set that turns off a second relay for the heater power (to act as a heat run-away safety), again once it turns on (which turns off the heater via the second relay in the series), it will not turn back off once the temp drops below 75C.
I also tried installing "Cooling" as a  way to do the above, but it conflicts and does not really work.

Or..........  could you perhaps put a "second", temperature in the ENC G-code command that lets say I am setting my Temp alarm 1 (fan cooling) to turn on at 72 degrees, and have a second Temp command that then when the temp reached lets say 71 degrees it would turn that fan off.  i.e.  lets say my temp alarm is id 5,  ENC O5 S72 K71     (the K value would be "Kill", thus the fan would turn on at 72 degrees due to heater overshoot or the like, then once it brought the temp down to 71C (or lower, the Kill temp) it would turn off the gpio that controls that fan.

Also, would there be a way to add/install a PID loop on the heater output? Mine is currently Bang/Bang (on/off to heater unit), would there be a way to add a PID loop control on the heater output, (or is there already one in the software)? (currently driving a SSR to run my enclosure heater).

vitormhenrique, 2021-11-04:
The alarm is only a "shot once" thing,  you really don't need to have an exaust fan to bring the temperature down.

after few seconds after the heater is off it should cool down by it self.

That being said, it is possible to create the type of logic that you want on the future.

poppabear1000, 2021-11-05:
Yes I would like the ability to be able to bring the temperature down in stages at this lower temperature has dropped down to the next stage same number of seconds by the temperature little bit less excetera and to it reaches room temperature sure if that is needed but it is my worry on pela and other parts that if the cooling is to Rapid they may.Sent from my Verizon, Samsung Galaxy smartphone