vitormhenrique/OctoPrint-Enclosure #451: Add "Minimum temperature" for cooling the enclosure on heat days.

al21883, 2021-11-26:
On the menu: "add output" -> "temperature/Humidity"
As there is a maximum temperature for "heater temp control type" in order to disable the heater once reached this temperature. It would be nice to have a **minimum temperature parameter** in order to disable the cooler once lowered the enclosure temperature to this target temp.

saxontown, 2022-01-08:
I have the same issue, a temp can be set for heater and works well, but i need to set a temp for an enclosure extractor fan for cooling so it turns on at a specific temp, and there is no temp setting option which i dont understand? Can this be added?
As an alternative i tried the Temperature Alarm option, but that also doesnt work because the GPIO selection dropdown only shows a black box instead of the GPIO pins. i see someone else has reported this issue in another ticket.
Right now its impossible to use this plugin to activate a cooling fan at a set temp. Im sure thousands of people would use it, if the cooling function in either function was working. PLEASE can this be resolved, would really be appreciated, and i would be happy to promote the plugin if these issues can be fixed soon. Thanks.

MTNBERG, 2022-03-06:
I'm glad to see that I'm not the only one looking at this problem. I currently have my enclosure fan setup to turn on when the pi CPU reaches a set temperature, but its done under an alarm setting which doesn't automatically shut the fan off once the temperature drops.
To me the temperature control option should have temperature set points for cooling as well.

[image]
.