vitormhenrique/OctoPrint-Enclosure #452: DHT22 not working for some time now

DarkDabula, 2021-11-29:
**Describe the bug**
Connected and working DHT22 reports no values inside OctoPrint. It worked before since some months ago.

**To Reproduce**
- Temperature and humidity show 0°C and 0%

**Expected behavior**
- Showing correct values of sensor. 

**Additional context**
Using OctoPrint 1.7.2 on a Rpi with Python 3.7. Getting errors in logfile

```
2021-11-29 18:03:13,123 - octoprint.server - INFO - ------------------------------------------------------------------------------
2021-11-29 18:03:13,200 - octoprint.plugins.enclosure - INFO - Failed to execute python scripts, try disabling use SUDO on advanced section of the plugin.
2021-11-29 18:03:13,201 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('not enough values to unpack (expected 2, got 1)',)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 1113, in read_dht_temp
    temp, hum = stdout.decode("utf-8").split("|")
ValueError: not enough values to unpack (expected 2, got 1)
```
After debugging a little bit, I made some slight changes to the file "getDHTTemp.py" at two locations

```
...
import sys
# Change: case-sensitive
import Adafruit_DHT as adafruit_dht

...

# Another way to read data
humidity, temperature = adafruit_dht.read(sensor, pin)

...
```

With this fix al works fine again.  Don't know what the problem is, maybe something with the Adafruit_DHT library.

Regards.

Albertowue, 2021-11-30:
Look #448, the adafruit library has changed

DarkDabula, 2021-11-30:
Hi,
Thanks for the hint. I read #448 but I think its a little bit misleading. There "Adafruit_DHT" is declared as the new API and "adafruit_dht" as the old one. More confusing is that you also can use "Adafruit_DHT" with Python 3.

Whatever ... After installing the "adafruit_dht" with the instructions from [Link](https://learn.adafruit.com/dht-humidity-sensing-on-raspberry-pi-with-gdocs-logging/python-setup) I got the original version of "getDHTTemp.py" working.

Thanks

closed 2021-11-30 by DarkDabula

Albertowue, 2021-11-30:
You are absolutely right, I corrected my post to create less confusion 😅