vitormhenrique/OctoPrint-Enclosure #453: No Sensor-readings, Variable "airquality" not declared for DHT11, DHT22, AM2302 ?

manuelwolfcgm, 2021-12-10:
**Describe the bug**
No Readings for DHT11 (which is tested and had been working up to the latest push) and AM2302. The sensors are working fine and already have been working in the enclosure-plugin. I needed to reinstall the plugin - and after that, I cant get any readings any more, although the sensor is accesible via a test-script.

**To Reproduce**
Fresh Plugin-Installation; Sensor shows no values; Log shows errors complaining about modifying a not-declared variable "airquality" in getDHTTemp.py

**Expected behavior**
....sensor should display values! (Tested via a smal test-py-script)

**Additional context**
[image]

It seems that in the last push, there have been adjustments in getDHTTemp.py, especially adding handling of the airquality-variable. For Sensors DHT11, 22, AM2302, this variable isn't declared, is it? I'll try so switch line 1026 to its previous version, not using the airqualtity-parameter (which is not useful for me). Maybe this is enough change in my case...
[image]
[image]

manuelwolfcgm, 2021-12-10:
....added
"airquality = 0"
in the if-clause for DHT11,22,2302 ( __init__.py )-> seems to work so far....
[image]
[image]

plani-de, 2021-12-11:
I had the same issue and adding the missing `airquality = 0` line helped at first, but now I continously get the follwoing error message:
`2021-12-11 20:47:18,056 - octoprint.plugins.enclosure - INFO - Failed to execute python scripts, try disabling use SUDO on advanced section of the plugin.

2021-12-11 20:47:18,057 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments: ('not enough values to unpack (expected 2, got 1)',)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 1127, in read_dht_temp
    temp, hum = stdout.decode("utf-8").split("|")
ValueError: not enough values to unpack (expected 2, got 1)
`

Anybody here who has an Idea. By the way, it makes no difference wether or not I disable/enable the "use SUDO". 
I also activated the logging function, but I get no futher details, e.g. what the AM2302 reads.

tanant, 2022-02-17:
@plani-de  not sure if it's the same cause however I'm talking through on #459 something that has the same error msg - I've put a rough summary of two things that I think are worth checking https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/459#issuecomment-1038530919 if that helps any?

tanant, 2022-02-18:
Gotcha, so yeah, that might mean this issue is resolved by #461, which is good to know

vitormhenrique, 2022-02-28:
Fixed!

closed 2022-02-28 by vitormhenrique