vitormhenrique/OctoPrint-Enclosure #456: Plugin failing to install due to RPi.GPIO issue

supawiz6991, 2021-12-16:
Environment:
RPi4 4GB
OS: Raspberry Pi OS Bullseye
Octoprint: Manual install due to Bullseye
  
**Describe the bug**
Plugin fails to install due to an error with RPi.GPIO

**To Reproduce**
Steps to reproduce the behavior:
1. Go to plugin manager
2. search for the plugin and click install
3. Installation will fail at RPi.GPIO

**Expected behavior**
Plugin was expected to install.

**Additional context**
I originally ran into the failure (see Stack Trace Error at the bottom of this issue)  while restoring an octoprint backup to the new installation. I removed enclosure plugin from the plugins_list.json and attempted the restore again and was successful. 

Post Restore, I went into the plugin manager and attempted to install the plugin from there and again got a failure notice.

In both cases the failure occurred during the installation of RPi.GPIO. I attempted to install RPi.GPIO via pip 

``pip install RPi.GPIO``

which errored out and failed.

The work around that fixed it (courtesy of jneilliii in the OcotoPrint discord) was doing:

``pip install RPi.GPIO==0.7.1a4``

Then upon attempting install of the plugin again it was successful. 

Stack Trace Error mentioned above:

`` Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/lib/python3.9/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 900, in _command_install_archive
    returncode, stdout, stderr = self._call_pip(pip_args)
  File "/home/pi/OctoPrint/venv/lib/python3.9/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 1631, in _call_pip
    return self._pip_caller.execute(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/lib/python3.9/site-packages/octoprint/util/pip.py", line 208, in execute
    return self.call(command, **kwargs)
  File "/home/pi/OctoPrint/venv/lib/python3.9/site-packages/octoprint/util/commandline.py", line 243, in call
    all_stderr += process_stderr(p.stderr.readlines(timeout=0.5))
  File "/home/pi/OctoPrint/venv/lib/python3.9/site-packages/octoprint/util/commandline.py", line 239, in process_stderr
    return process_lines(lines, self._log_stderr)
  File "/home/pi/OctoPrint/venv/lib/python3.9/site-packages/octoprint/util/commandline.py", line 229, in process_lines
    processed = self._preprocess_lines(
  File "/home/pi/OctoPrint/venv/lib/python3.9/site-packages/octoprint/util/pip.py", line 498, in _preprocess_lines
    return list(map(self._preprocess, lines))
  File "/home/pi/OctoPrint/venv/lib/python3.9/site-packages/octoprint/util/pip.py", line 517, in _preprocess
    return to_unicode(clean_ansi(text))
  File "/home/pi/OctoPrint/venv/lib/python3.9/site-packages/octoprint/util/commandline.py", line 57, in clean_ansi
    return _ANSI_REGEX.sub(b"", line.encode("latin1")).decode("latin1")
UnicodeEncodeError: 'latin-1' codec can't encode character '\u2018' in position 34: ordinal not in range(256) ``

bullitt186, 2021-12-19:
Thank you! This solved exactly my problem :-)
Worth mentioning: you have to activate your python octoprint virtual environment. In case of a default OctoPi Installation, this would be:

`source ~/oprint/bin/activate
` 
(given that you are logged in as User "pi")

supawiz6991, 2022-02-12:
Update on this:
The developer of RPi.GPIO has released full version 0.7.1 into the repository which includes the fix from the alpha. This means the work around is no longer needed.

That being said, there is a bug in pip 22.0.3 which causes the plugin to fail to install. Work around is to downgrade pip to 21.3.1.

vitormhenrique, 2022-02-28:
I'm guessing this is out of my control, hopefully pip and the gpio library will play nice in a near future. but thank you for documenting the work around and how to fix it!
gonna change the title to make it easy for people to find this

closed 2022-02-28 by vitormhenrique