vitormhenrique/OctoPrint-Enclosure #457: DHT22 am2302 no values

schlicr, 2021-12-22:
Hello Guys, 
after a while of researching, I'm still not able to get the Enclosure Plugin to run.

It alwasy fails with the Messages underneath.

My Enviorment is: Octoprint Version 1.7.2
I can get Vaules from my AM2302 when I do it like that:

_>> python3  getDHTTemp.py 22 2
14.8 | 40.5_

If I do it like this one: 

_>> python  getDHTTemp.py 22 2
Traceback (most recent call last):
  File "getDHTTemp.py", line 2, in <module>
    import adafruit_dht
ImportError: No module named adafruit_dht_

It Fails. so what is happend here? Does anyone how to fix that? Any help would be apriciated.

Thanks.

Two Python versions:
Python 2.7.16
Python 3.7.3

Here are some messages from octoprint.log:

_octoprint.plugins.enclosure - WARNING - An exception of type UnboundLocalError occurred on log_error. Arguments:
("local variable 'airquality' referenced before assignment",)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 1026, in get_sensor_data
    if temp != -1 and hum != -1 and airquality != -1:
UnboundLocalError: local variable 'airquality' referenced before assignment
octoprint.plugins.enclosure - WARNING - An exception of type TypeError occurred on log_error. Arguments:
('cannot unpack non-iterable NoneType object',)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 811, in check_enclosure_temp
    temp, hum, airquality = self.get_sensor_data(sensor)
TypeError: cannot unpack non-iterable NoneType object_

schlicr, 2021-12-22:
Oh, I fixed that Version thing.
_python getDHTTemp.py 22 2_

Gives me an result now. But unfortunatly Enclosure Plugin gives me the same errors

schlicr, 2021-12-22:
This are the messages I get, when I disable "SUDO" in advaced options.

?????

2021-12-22 18:27:22,391 - octoprint.plugins.enclosure - WARNING - An exception of type UnboundLocalError occurred on log_error. Arguments:
("local variable 'airquality' referenced before assignment",)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 1026, in get_sensor_data
    if temp != -1 and hum != -1 and airquality != -1:
UnboundLocalError: local variable 'airquality' referenced before assignment
2021-12-22 18:27:22,391 - octoprint.plugins.enclosure - WARNING - An exception of type TypeError occurred on log_error. Arguments:
('cannot unpack non-iterable NoneType object',)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 811, in check_enclosure_temp
    temp, hum, airquality = self.get_sensor_data(sensor)
TypeError: cannot unpack non-iterable NoneType object

webportpl, 2022-02-01:
Unfortunatly Enclosure Plugin on dashboard don't have values :(