vitormhenrique/OctoPrint-Enclosure #458: Impossible to install Enclosure pluggin

facoco83, 2021-12-23:
**Describe the bug**
Hi, I have installed Octoprint on a RPI3B+ today. 
I want to add the pluggin OctoPrint-Enclosure but when I want to do it by the Octoprint UI, I have this message : 
`>OctoPrint->OctoPrint-Enclosure==4.13.2) (0.1.7)
Using legacy 'setup.py install' for OctoPrint-Enclosure, since package 'wheel' is not installed.
Using legacy 'setup.py install' for RPi.GPIO, since package 'wheel' is not installed.
Installing collected packages: RPi.GPIO, RPi.bme280, gpiozero, bme680, OctoPrint-Enclosure
Running setup.py install for RPi.GPIO: started
Running setup.py install for RPi.GPIO: finished with status 'error'
Error!
Could not install plugin from https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/master.zip, see the log for more details` at the end of the logs of installation. When I go into the Log file, I have not more information. It's the same result. 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Plungin Manager --> +Get More --> 'Octoprint-Enclosure
2. Click on 'Install'
3. See error

**Expected behavior**
Can you help me please? 
I has installed a old version on Octoprint with the old version of Enclosure plugin but without problem. The old version of Enclosure is not available.

fietser28, 2021-12-23:
I had a similar problem after upgrading a raspberry pi (raspbian) to bullseye which has python3.9. I fixed this with:

1. Change into the venv of you octoprint installation, command (note the dot, replace <venvpath> with you path):  
     . <venvpath>/bin/activate)
2. Run: pip3 install RPi.GPIO==0.7.1a4
3. Restart octoprint
4. Install enclosure plugin from octoprint UI, it now works.

For the time being maybe add this version dependency to the enclosure plugin?
It is a problem with RPi.GPIO, not this plugin. Also see: https://sourceforge.net/p/raspberry-gpio-python/tickets/198/
The release cycle of RPi.GPIO doesn't look very fast on sourceforge.

facoco83, 2021-12-24:
Thank you, I have prepared a new raspbian for do a new install. 
When I have installed octoprint, I have installed pip3 install RPi.GPIO==0.7.1a4 in the venv and after I have succeed to install the Pluggin Enclosure. 

Joyeuses fêtes à tous

burbilog, 2022-01-19:
I had the same issue when I tried to install this plugin in my brand new octoprint instance. Thanks for the fix, it worked.

vitormhenrique, 2022-02-28:
documented here: https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/456

closed 2022-02-28 by vitormhenrique