vitormhenrique/OctoPrint-Enclosure #459: Can't get DHT11 Sensor to work

laurensguijt, 2021-12-28:
Hello guys,

After trying pretty much everything I still can't get the sensor temp to display. I have already tried disabling the use SUDO box. The libraries are installed and seem to be working correctly according to the testing script. See attached images underneath: 

-Laurens 

[image]
[image]
[image]
[image]

tanant, 2022-01-13:
I had a spare DHT11 so I connected it up, got the same issue, so will have a quick play with it this weekend if I get some time but to chase down a hunch @DeathmannX  / @laurensguijt - in your cases, are you running your octoprint server under the `pi` username? (I'm not, but it'll just mean one less thing to worry about in your cases)

I'm guessing it's this:
- if you run this without `sudo` it should run as the `pi` user and have access to the `adafruit_dht` library _but_ you need to be root to access the GPIO chip (I think)
- if you run it _with_ `sudo`, then you have access to the GPIO chip, but you're running as root and won't have access to the relevant libraries.

if you want to give that a shot @laurensguijt - try the dht_simpletest but run with sudo, do you get something like this?

```
pi@octopi:/home/octo $ sudo python3 ~/foo.py
Traceback (most recent call last):
  File "/home/pi/foo.py", line 5, in <module>
    import board
ModuleNotFoundError: No module named 'board'
```

F3n1x040, 2022-02-13:
hi I have the same problem, the sensor works well with the example in Python, but no data appears in the octoprint interface. it also happens to me with the CPU temperature sensor. thanks in advance for the help

tanant, 2022-02-14:
heya @F3n1x040 (and anyone else watching) Lemme know if this works and I'll do a documentation PR at the very least.. There are two things I had to make sure of. 

1. The [Adafruit CircuitPython DHT python module](https://github.com/adafruit/Adafruit_CircuitPython_DHT) is accessible to the user that is running octoprint
2. The user that is running needs to have access to the GPIO pins. TL;DR - probably needs to be part of the _gpio_ group. (see note below though!)

Toggling sudo would solve the second, but would cause the first issue. If you have the library installed, but you lack access you'll see something like `... - octoprint.plugins.enclosure - DEBUG - Dht result: b'Unable to open chip: gpiochip0\n'` 

### If you need to run with sudo?
1. Install the module system-wide.
```pi@blueberry_crumble:~ $ sudo pip3 install adafruit-circuitpython-dht ```
2. Enclosure should start picking up things now (you might need to reboot Octoprint, i don't think so but maybe. I forget)

### If you don't need sudo, or don't want to run with it? (what i'm doing here)
1. In the terminal as the user that is running octoprint (`pi` in your case) install the module
`pi@blueberry_crumble:~ $ pip3 install adafruit-circuitpython-dht`
2. then **add the user pi to the gpio group** (or dialout, see below)
`pi@blueberry_crumble:~ $ sudo usermod --append --groups gpio pi`
3. At this point you'll probably need to reboot the RPi.

## gpio or dialout?
I did find some links suggesting you need to be part of the _dialout_ group - https://askubuntu.com/a/1313156 but on my RPi it's not dialout, it's gpio. Here's how I double checked:
```
pi@blueberry_crumble:~ $ less /etc/udev/rules.d/99-com.rules
...
...
SUBSYSTEM=="gpio", GROUP="gpio", MODE="0660"
SUBSYSTEM=="gpio", KERNEL=="gpiochip*", ACTION=="add", PROGRAM="/bin/sh -c 'chgrp -R gpio /sys/class/gpio && chmod -R g=u /sys/class/gpio'"
SUBSYSTEM=="gpio", ACTION=="add", PROGRAM="/bin/sh -c 'chgrp -R gpio /sys%p && chmod -R g=u /sys%p'"
...
...
```
I don't recall doing that, and that udev config is the same on my other RPi I use for a local DNS proxy.

F3n1x040, 2022-02-14:
hello @tanant, I explain, I have verified that the user who runs octoprint is "pi" and that "pi" has access to GPIO pins.
`pi@octopi:~/oprint/lib/python3.7/site-packages/octoprint_enclosure $ groups pi
pi : pi adm tty dialout cdrom sudo audio video plugdev games users input netdev spi i2c gpio`

if I run the command "systemctl status octoprint" I get the following:
`pi@octopi:~/oprint/lib/python3.7/site-packages/octoprint_enclosure $ systemctl status octoprint`
`● octoprint.service - The snappy web interface for your 3D printer`
 `  Loaded: loaded (/etc/systemd/system/octoprint.service; enabled; vendor preset: enabled)`
`   Active: active (running) since Mon 2022-02-14 19:40:23 CET; 12min ago`
 `Main PID: 1304 (octoprint)`
 `   Tasks: 25 (limit: 1935)`
  ` CGroup: /system.slice/octoprint.service`
 `          └─1304 /home/pi/oprint/bin/python3 /home/pi/oprint/bin/octoprint serve --host=127.0.0.1 --port=5000`

`Feb 14 19:52:39 octopi octoprint[1304]: Traceback (most recent call last):
Feb 14 19:52:39 octopi octoprint[1304]:   File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__
Feb 14 19:52:39 octopi octoprint[1304]:     if temp != -1 and hum != -1 and airquality != -1:
Feb 14 19:52:39 octopi octoprint[1304]: UnboundLocalError: local variable 'airquality' referenced before assignment
Feb 14 19:52:39 octopi octoprint[1304]: 2022-02-14 19:52:39,633 - octoprint.plugins.enclosure - WARNING - An exception o
Feb 14 19:52:39 octopi octoprint[1304]: ('cannot unpack non-iterable NoneType object',)
Feb 14 19:52:39 octopi octoprint[1304]: Traceback (most recent call last):
Feb 14 19:52:39 octopi octoprint[1304]:   File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__
Feb 14 19:52:39 octopi octoprint[1304]:     temp, hum, airquality = self.get_sensor_data(sensor)
Feb 14 19:52:39 octopi octoprint[1304]: TypeError: cannot unpack non-iterable NoneType object
lines 1-18/18 (END)`
From which I deduce that something happens with the variable "airquality"

When looking at the Octoprint log, the following warning and error lines appear:
`2022-02-14 17:52:48,478 - octoprint.plugins.enclosure - INFO - Failed to execute python scripts, try disabling use SUDO on advanced section of the plugin.
2022-02-14 17:52:48,478 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('not enough values to unpack (expected 2, got 1)',)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 1126, in read_dht_temp
    temp, hum = stdout.decode("utf-8").split("|")
ValueError: not enough values to unpack (expected 2, got 1)
2022-02-14 17:52:48,479 - octoprint.plugins.enclosure - WARNING - An exception of type UnboundLocalError occurred on log_error. Arguments:
("local variable 'airquality' referenced before assignment",)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 1026, in get_sensor_data
    if temp != -1 and hum != -1 and airquality != -1:
UnboundLocalError: local variable 'airquality' referenced before assignment
2022-02-14 17:52:48,479 - octoprint.plugins.enclosure - WARNING - An exception of type TypeError occurred on log_error. Arguments:
('cannot unpack non-iterable NoneType object',)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 811, in check_enclosure_temp
    temp, hum, airquality = self.get_sensor_data(sensor)
TypeError: cannot unpack non-iterable NoneType object`

I'm using the plugin without "sudo", but I've tried using it with "sudo" and the log doesn't change
I have also installed the system-wide module both with the sudo command, and without the

I have tried to assign a value to the variable "airquality" in the code of the file "__init__.py" in the following line:

> def get_sensor_data(self, sensor):
        try:
            if self.development_mode:
                temp, hum, airquality = self.read_dummy_temp()
            else:
                if sensor['temp_sensor_type'] in ["11", "22", "2302"]:

airquality = 0
>  temp, hum = self.read_dht_temp(sensor['temp_sensor_type'], sensor['gpio_pin'])
                elif sensor['temp_sensor_type'] == "18b20":
                    temp = self.read_18b20_temp(sensor['ds18b20_serial'])

But this causes the plugin not to load

tanant, 2022-02-17:
> From which I deduce that something happens with the variable "airquality"

Good spot, I'm not actually sure how it ends up there so we can look at it in a sec once we crush the first error that comes in a lot earlier - i have a suspicion that it's symptomatic rather than the thing we want to fix which is this:

```
File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 1126, in read_dht_temp temp, hum = stdout.decode("utf-8").split("|")
ValueError: not enough values to unpack (expected 2, got 1)
```
That suggests you're having trouble reading data from the sensor, the rest of the issues sort of cascade from that. To isolate this down, and take enclosure out of the picture, could you do the following?

- as the user `pi` try run the `getDHTTemp.py` script directly (it lives alongside the `__init__.py` file if you hadn't found it)

```
pi@blueberry_crumble: $ cd /home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure
pi@blueberry_crumble: $ python3 getDHTTemp.py 11 4
21.0 | 38.0
```
The 11 is for a DHT11 (22 is for a DHT22). 
The 4 is which GPIO pin it's on.

If you supply a bad pin OR the wrong sensor then odds on it'll report `-1 | -1`. 
If there's something wrong with your setup then this will error out. It'll either complain about an inability to load the `adafruit_dht` module OR that it can't open `gpiochip0`. 

If it's the adafruit_dht module, it's the first step i mentioned. 
If it's the gpiochip, then something in your setup is preventing access to the chip.
If it reads back correct data, then something _else_ i'm not describing is happening - but we will know at least a portion of your setup is good.

(also, if the edits you've made cause the plugin to not load - I'm guessing the python you inserted has mixed spaces and tabs or has an indentation issue? Just a gut feeling.)

tanant, 2022-02-18:
Ah, awesome, so does everything now work and you're getting proper readouts in enclosure? If so, was did you end up needing to change from a standard install?

F3n1x040, 2022-02-18:
I don't know what to answer you. I did a standard installation with the octoprint manager plugin, then followed the steps indicated for the DHT11, DHT22 and AM2302 sensors as indicated by the readme.md and in the end I had to modify the file `__init__.py` adding the line `airquality = 0` as described by @manuelwolfcgm in issue #453
So it could be said that I did have to modify the standard installation

F3n1x040, 2022-02-19:
hello, things that happen hehe. the good thing you already know is what has happened.
My thanks for your plugin and your work as well.
Best regards