vitormhenrique/OctoPrint-Enclosure #462: Adafruit BME680 Temperature sensor not connecting to Enclosure Plugin

BSBarrows, 2021-12-31:
I have a Raspberry Pi 4 wired up to an Adafruit BME680 board over the 4 cable quick connector, connected like this 
[image]
But I am not able to get it to work with the Enclosure plugin. I believe I would set the Enclosure plugin for GPIO 2 or 3 but any time I change it from 4 I am not able to run: i2cdetect -y 1 without it taking forever or failing.
[image]
I believe the BME680.py script runs correctly, and this is with GPIO set to 4, I can run i2cdetect and it quickly returns 0x77.
[image]

Any help would be great, I would love to be able to see the Temp and Humidity in my Enclosure and on my Plotty graph.

BSBarrows, 2022-01-09:
If I select logging, this is what I found in the octoprint log:

2022-01-09 10:46:24,700 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('not enough values to unpack (expected 3, got 1)',)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 1177, in read_bme680_temp
    temp, hum, airq = output.split("|")
ValueError: not enough values to unpack (expected 3, got 1)
2022-01-09 10:46:24,700 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('not enough values to unpack (expected 3, got 2)',)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 993, in get_sensor_data
    temp, hum, airquality = self.read_bme680_temp(sensor['temp_sensor_address'])
ValueError: not enough values to unpack (expected 3, got 2)
2022-01-09 10:46:24,700 - octoprint.plugins.enclosure - WARNING - An exception of type TypeError occurred on log_error. Arguments:
('cannot unpack non-iterable NoneType object',)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 811, in check_enclosure_temp
    temp, hum, airquality = self.get_sensor_data(sensor)
TypeError: cannot unpack non-iterable NoneType object

BSBarrows, 2022-01-09:
[octoprint-systeminfo-20220109120332.zip](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/7835510/octoprint-systeminfo-20220109120332.zip)
I have attached the Octoprint System Info bundle now as well in case that helps.

n8many, 2022-02-23:
Hi,

I think I have this fixed in #482. Can you test to make sure the changes in that pull request resolve this for you?

Thanks