vitormhenrique/OctoPrint-Enclosure #464: Temperature Alarm no option on controlled IO

Steve1987113, 2022-01-05:
All other outputs work fine and can add the gpio number. On temperature alarm output it wont allow me to input a controlled IO input.
[image]

saxontown, 2022-01-08:
I have the same issue, doesnt seem to be any way to add controlled cooling via temp alarm, or the temp/humidity control functions. PLEASE PLEASE PLEASE can this be fixed. Thank you.

Braunc55, 2022-01-25:
I may be misunderstanding the issue but the trigger for the Temperature Alarm output is a Temperature Sensor (defined as an Enclosure Input), then a Regular IO Enclosure Output needs to be defined and selected as the Temperature Alarm Controlled IO. Does that solve the issue?

Steve1987113, 2022-01-25:
> I may be misunderstanding the issue but the trigger for the Temperature Alarm output is a Temperature Sensor (defined as an Enclosure Input), then a Regular IO Enclosure Output needs to be defined and selected as the Temperature Alarm Controlled IO. Does that solve the issue?

I have my temp sensor selected under  temp sensor (enclosure temp) but when you go to select the controlled IO under the drop down menu there is nothing there, all you get is half a missing box like the the webpage is broken if you know what i mean, if you have a closer look at the screen shot you can see ive clicked the drop down menu and there is just a line that appears.

Braunc55, 2022-01-25:
> I have my temp sensor selected under temp sensor (enclosure temp) but when you go to select the controlled IO under the drop down menu there is nothing there, all you get is half a missing box like [...]

Once you define a Regular IO output it will appear in the controlled IO drop down.

saxontown, 2022-01-26:
Thanks for the replies. Sorry i'm still not understanding, this is, my input and output screenshots https://www.screencast.com/t/kiBKewYrg2 and https://www.screencast.com/t/ZKRFkgftP but the Controlled IO is not selectable. I cant see what else to change. Do the screens help?

Steve1987113, 2022-01-26:
> Thanks for the replies. Sorry i'm still not understanding, this is, my input and output screenshots https://www.screencast.com/t/kiBKewYrg2 and https://www.screencast.com/t/ZKRFkgftP but the Controlled IO is not selectable. I cant see what else to change. Do the screens help?

I think i know what the problem is now, first you need to create a regular IO output as in screen shot
[image]
Then you have to create a Temperature alarm Output and select the Controlled IO what you labelled 
[image]

saxontown, 2022-01-26:
I really appreciate your time and quick responses. Please see this video, i'm obviously still not getting it. Is there something really obvious i have missed? https://www.screencast.com/t/y9a5fPTQw

saxontown, 2022-01-26:
Thank you for staying with me on this. I have it working, just as you explained. The IO turns on as expected at the set temperature. Question, when the temp drops below the set temp, should the the IO port go high and turn off the GPIO? Asking as i notice the GPIO output stays on once triggered. Or is it the case its an alarm and shouldnt reset if the temp drops?

Second question, do you know anything about the Temperature/Humidity Control option? Assuming the Temp Alarm function doesnt unlatch once activated, i was thinking the temp/humidity control might work better (to use a extractor fan for cooling - going on/off as required to maintain a temperature) however i notice once the 'cooler' option is set in this module, the temperature field disappears (its there for the heater option only). Any idea how this temp/humidity module can be set for cooling with a max temperature value? https://www.screencast.com/t/zF0U0x4fK75p

A big thank you for the assistance and fast responses.

closed 2022-02-26 by Steve1987113