vitormhenrique/OctoPrint-Enclosure #471: Sensor Si7021 not working

Dexter0007, 2022-01-18:
Hi, I have a problem with temperature sensor.
The sensor was detected on 0x40 address but enclosure plugin showing only 0.
[image]
Disable SUDO not helped, sudo without password.

Octoprint log:
`2022-01-18 17:20:20,933 - octoprint.plugins.enclosure - INFO - Failed to execute python scripts, try disabling use SUDO on advanced section of the plugin.
2022-01-18 17:20:20,933 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('not enough values to unpack (expected 2, got 1)',)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 1283, in read_si7021_temp
    temp, hum = stdout.decode("utf-8").split("|")
ValueError: not enough values to unpack (expected 2, got 1)`

OctoPrint 1.7.2 Python 3.7.3 OctoPi 0.18.0

grahamwoan, 2022-03-11:
yes, same for me

Cookietyrant, 2022-05-27:
I struggled with this all night but somehow got it to work with a shotgun approach...not sure if the below worked, but I did also at the end try disabling root and then the readings started coming

sudo apt-get install build-essential libi2c-dev i2c-tools python-dev libffi-dev
sudo -H pip install cffi 
sudo -H pip install smbus-cffi 
sudo -H pip install Si7021
[image]
[image]

jx2014, 2022-06-30:
I had the same problem.
~/.octoprint/logs/octoprint.log
repeatedly showing the following tracebacks:

`2022-06-21 15:27:11,691 - octoprint.plugins.enclosure - WARNING - An exception of type KeyError occurred on log_error. Arguments:
('temp_sensor_i2cbus',)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 996, in get_sensor_data
    temp, hum = self.read_si7021_temp(sensor['temp_sensor_address'], sensor['temp_sensor_i2cbus'])
KeyError: 'temp_sensor_i2cbus'
2022-06-21 15:27:11,692 - octoprint.plugins.enclosure - WARNING - An exception of type TypeError occurred on log_error. Arguments:
('cannot unpack non-iterable NoneType object',)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 811, in check_enclosure_temp
    temp, hum = self.get_sensor_data(sensor)
TypeError: cannot unpack non-iterable NoneType object
2022-06-21 15:27:12,352 - octoprint.plugins.tracking - INFO - Sent tracking event pong, payload: {'version': '1.8.1', 'os': 'linux', 'bits': 32, 'python': '3.7.3', 'pip': '21.3.1', 'cores': 4, 'freq': 1200.0, 'ram': 915714048, 'pi_model': 'Raspberry Pi 3 Model B Rev 1.2', 'octopi_version': '0.17.0', 'plugins': 'obico:2.0.1,firmwareupdater:1.13.3,enclosure:4.13.2,filamentsensorsimplified:0.3.1,ender3v2tempfix:0.0.4,bedlevelvisualizer:1.1.1'}
2022-06-21 15:27:21,693 - octoprint.plugins.enclosure - WARNING - An exception of type KeyError occurred on log_error. Arguments:
('temp_sensor_i2cbus',)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 996, in get_sensor_data
    temp, hum = self.read_si7021_temp(sensor['temp_sensor_address'], sensor['temp_sensor_i2cbus'])
KeyError: 'temp_sensor_i2cbus'
2022-06-21 15:27:21,694 - octoprint.plugins.enclosure - WARNING - An exception of type TypeError occurred on log_error. Arguments:
('cannot unpack non-iterable NoneType object',)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 811, in check_enclosure_temp
    temp, hum = self.get_sensor_data(sensor)
TypeError: cannot unpack non-iterable NoneType object`

As a temporary solution, make sure the sensor actually works by going to
`~/oprint/lib/python3.7/site-packages/octoprint_enclosure`

Manually run the sensor to see if you can read the temperature and humidity properly:
```
pi@octopi:~/oprint/lib/python3.7/site-packages/octoprint_enclosure $ python3 SI7021.py 40 1
41.0 | 28.2
```
Note **40** is the i2c address, **1** is the i2c bus.

Next, open **__init__.py** and manually hardcode the **temp_sensor_address** and **temp_sensor_i2cbus** to line 996
This is the original:
`temp, hum = self.read_si7021_temp(sensor['temp_sensor_address'], sensor['temp_sensor_i2cbus'])`
change it to:
`temp, hum = self.read_si7021_temp(40, 1)`

Reboot octoprint, it should now be able to read the temp and humidity.