vitormhenrique/OctoPrint-Enclosure #475: Control a 12v LED Strip with the software

110Tim706, 2022-02-02:
Hey,

I'm kinda confused. First this ISN'T mainly a feature request. I'd like to turn off my 12V LED Strip (just + and - no data, because just white) and I don't know how to do it and how to add in the software.

I thought it should be possible to control the Strip with a relais. But I don't know which kind of relais I need and how I have to connect it.

PS: I'm new to electronics and Octoprint.

Already thanks to everyone who thinks about my problem. If you have any ideas just comment maybe this will gonna help other people too.

tankdeer, 2022-02-03:
This is actually very easy with a MOSFET or relay. The benefit of a MOSFET is you can define the GPIO as PWM and then not only turn on/off, but adjust brightness as well. This guide should help. It's for RGB, but the same logic applies - you just apply it to the ground instead of the specific color channel. 1/3 the work.

https://dordnung.de/raspberrypi-ledstrip/

110Tim706, 2022-02-03:
> This is actually very easy with a MOSFET<

Is this MOSFET typ okay? IRLZ34N

tankdeer, 2022-02-03:
> > This is actually very easy with a MOSFET<
> 
> Is this MOSFET typ okay? IRLZ34N

Yes, that is what I used

110Tim706, 2022-02-03:
> > > This is actually very easy with a MOSFET<
> > 
> > 
> > Is this MOSFET typ okay? IRLZ34N
> 
> Yes, that is what I used

Can you tell me how you connected it? With a sketch or so? Already thanks so much.😊

tankdeer, 2022-02-03:
I don't speak German, no. No more than a few words here and there at least. :) The tutorial has English and German available (There should be a link on the top to change language)

This image from the tutorial should give you a good idea of the wiring:

https://dordnung.de/raspberrypi-ledstrip/img/rgb/small/power_3.jpg?ver=1.0

Basically it should be:

+12v power -> +12v on LED strip
Ground Power -> Ground pin on Raspberry pi _and_ Source pin (right) on MOSFET
Ground on LED Strip -> Drain pin (middle) on MOSFET
GPIO of your choice on rPi -> Source (left) pin on MOSFET

Make sure the two grounds are connected, and make sure you don't accidentally put 12v to the pi and you'll be fine.

110Tim706, 2022-02-03:
> I don't speak German, no. No more than a few words here and there at least. :) The tutorial has English and German available (There should be a link on the top to change language)
>
> This image from [...]

Thanks :)

tankdeer, 2022-02-03:
No problem. :)

vitormhenrique, 2022-02-28:
As stated here, pretty easy to do. Not an issue

closed 2022-02-28 by vitormhenrique