vitormhenrique/OctoPrint-Enclosure #493: DHT22 Temp sensor on Pi 4

pman92, 2022-10-15:
Hey guys,
I've recently changed from using a pi 3b+ to a pi 4b.
I previously had a DHT22 temp sensor working with the enclosure plugin on the 3B+. Its now connected the same (to GPIO 4) on the 4B but I cannot get it to work. 

1. I have SSH'ed and installed the adafruit-circuitpython-dht and libgpiod2 as described in the enclosure plugin readme. 
2. I have tried disabling "use SUDO" in enclosure plugin advanced settings (as I had to do that to get it working on the 3B+, but it made no difference on the 4B). 
3. I have tried the adafruit sample code, and get the following:
`pi@octopi:~ $ python3 dht_simpletest.py
Traceback (most recent call last):
  File "dht_simpletest.py", line 5, in <module>
    import board
  File "/home/pi/.local/lib/python3.7/site-packages/board.py", line 44, in <module>
    from adafruit_blinka.board.raspberrypi.raspi_4b import *
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_blinka/board/raspberrypi/raspi_4b.py", line 6, in <module>
    from adafruit_blinka.microcontroller.bcm2711 import pin
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_blinka/microcontroller/bcm2711/pin.py", line 5, in <module>
    from RPi import GPIO
ModuleNotFoundError: No module named 'RPi'
`

pman92, 2022-10-15:
I fixed the "no module found" error by running "sudo apt install python3-rpi.gpio". 
However the sensor still doesn't work, I now get "DHT sensor not found, check wiring" from the adafruit test script.

The sensor is connected to GPIO4 (pin 7). Inside the script I have:
`dhtDevice = adafruit_dht.DHT22(board.D4)`
and have also tried:
`dhtDevice = adafruit_dht.DHT22(board.D4, use_pulseio=False)`

pman92, 2022-10-15:
Have got it working. 
After finding [this](https://lightrun.com/answers/adafruit-adafruit_circuitpython_dht-a-full-buffer-was-not-returned--try-again), I installed the following:
`pip3 install sysv-ipc`
And now the sensor is working fine in both octoprint and the test script. 

I still require "use SUDO" to be disabled under advanced settings like I did with the 3B+. Which makes sense because if a run the test script as sudo it still says sensor not found.

burnerbmx, 2023-07-02:
i had this issue but manged to get rounf that im using the dht11 sensor and its reading temp and humi but to dislaying the values on my octoprint