vitormhenrique/OctoPrint-Enclosure #494: Pi 4 DHT 22 Not working

stonemandave, 2022-10-28:
2022-10-28 15:58:37,936 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('not enough values to unpack (expected 2, got 1)',)
Traceback (most recent call last):
  File "/home/slacks/oprint/lib/python3.7/site-packages/octoprint_enclosure/__init__.py", line 1153, in read_dht_temp
    temp, hum = stdout.decode("utf-8").split("|")
ValueError: not enough values to unpack (expected 2, got 1)

This the log i got. The dht works i tested before and after. I don't need resister because mine is built in. I have it going to 5v.

schmidtflo, 2022-11-04:
Had the exact same issue, `pip3 install sysv-ipc` fixed it for me

stonemandave, 2022-11-05:
I forgot to mention I installed alot of different pips and it works now

stonemandave, 2022-11-05:
Hopefully this helps some else

closed 2022-11-05 by stonemandave