vitormhenrique/OctoPrint-Enclosure #500: installation issue

VolkerMandt, 2023-11-18:
**Success to install the plugin [OctoPrint-Enclosure], but no plugin**
Installation from list of the OctoPrint Plugin Repository, OctoPrint says success, after restart it s not visible, not in the list, not on the surface, 
When I install it once more it says old files uninstalled, and new installed successfully (problem not solved)
I tried to install per file and https, same issue. 
I can install other plugins. 
Looking for OctoPrint-Enclosure-V2, but there are no files. 

Do anyone have a clue? 

Greetings, 
Volker Mandt

VolkerMandt, 2023-11-18:
PS: I use a Pi 4B, the version of Octoprint is the latest...

PetrovskYYY, 2024-02-27:
Same issue. 
popup window
```
Plugin management log 

Install Unknown: 
A restart is needed for the changes to take effect.
```
Install log:
```
Installing plugin "OctoPrint-Enclosure" from https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/master.zip...
/home/sysadmin/OctoPrint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmp9mdl13i6/OctoPrint-Enclosure-master.zip --no-cache-dir
Processing /tmp/tmp9mdl13i6/OctoPrint-Enclosure-master.zip
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'done'
Requirement already satisfied: OctoPrint in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint-Enclosure==4.13.2) (1.9.3)
Requirement already satisfied: RPi.GPIO>=0.6.5 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint-Enclosure==4.13.2) (0.7.1)
Requirement already satisfied: RPi.bme280 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint-Enclosure==4.13.2) (0.2.4)
Requirement already satisfied: bme680 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint-Enclosure==4.13.2) (1.1.1)
Requirement already satisfied: gpiozero==1.6.2 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint-Enclosure==4.13.2) (1.6.2)
Requirement already satisfied: requests>=2.7 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint-Enclosure==4.13.2) (2.31.0)
Requirement already satisfied: smbus2>=0.3.0 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint-Enclosure==4.13.2) (0.4.3)
Requirement already satisfied: colorzero in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from gpiozero==1.6.2->OctoPrint-Enclosure==4.13.2) (2.0)
Requirement already satisfied: charset-normalizer<4,>=2 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from requests>=2.7->OctoPrint-Enclosure==4.13.2) (3.3.2)
Requirement already satisfied: idna<4,>=2.5 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from requests>=2.7->OctoPrint-Enclosure==4.13.2) (3.6)
Requirement already satisfied: urllib3<3,>=1.21.1 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from requests>=2.7->OctoPrint-Enclosure==4.13.2) (1.26.18)
Requirement already satisfied: certifi>=2017.4.17 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from requests>=2.7->OctoPrint-Enclosure==4.13.2) (2023.11.17)
Requirement already satisfied: smbus in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from bme680->OctoPrint-Enclosure==4.13.2) (1.1.post2)
Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.2) (2021.2.23)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.2) (2021.10.11)
Requirement already satisfied: OctoPrint-PiSupport>=2023.5.24 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.2) (2023.10.10)
Requirement already satisfied: argon2-cffi<22,>=21.3.0 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.2) (21.3.0)
Requirement already satisfied: Babel<2.13,>=2.12.1 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.2) (2.12.1)
Requirement already satisfied: cachelib<0.11,>=0.10.2 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.2) (0.10.2)
Requirement already satisfied: Click<9,>=8.1.3 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.2) (8.1.7)
Requirement already satisfied: colorlog<7,>=6.7.0 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.2) (6.8.0)
Requirement already satisfied: emoji<3,>=2.2.0 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.2) (2.9.0)
Requirement already satisfied: feedparser<7,>=6.0.10 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.2) (6.0.11)
Requirement already satisfied: filetype<2,>=1.2.0 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.2) (1.2.0)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.2) (2.1.0)
Requirement already satisfied: Flask-Babel<4,>=3.1.0 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.2) (3.1.0)
Requirement already satisfied: Flask-Login<0.7,>=0.6.2 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.2) (0.6.3)
Requirement already satisfied: Flask-Limiter<4,>=3.3.0 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.2) (3.5.0)
Requirement already satisfied: flask<2.3,>=2.2.3 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.2) (2.2.5)
Requirement already satisfied: frozendict<3,>=2.3.7 in /home/sysadmin/OctoPrint/lib/python3.11/site-packages (from OctoPrint->OctoPrint-Enclosure==4.13.2) (2.3.10)
[... cut ...]
Created wheel for OctoPrint-Enclosure: filename=OctoPrint_Enclosure-4.13.2-py3-none-any.whl size=73407 sha256=5e01fbe9b1235ce43817f860aa8078d36a025ccf23772cd51b850dbf80dc26b1
Stored in directory: /tmp/pip-ephem-wheel-cache-zrojcw1x/wheels/00/6d/f0/913e6ff67c34225fe6d21338b4b550e39e95f83e59d2965a6e
Successfully built OctoPrint-Enclosure
Installing collected packages: OctoPrint-Enclosure
Attempting uninstall: OctoPrint-Enclosure
Found existing installation: OctoPrint-Enclosure 4.13.2
Uninstalling OctoPrint-Enclosure-4.13.2:
Successfully uninstalled OctoPrint-Enclosure-4.13.2
Successfully installed OctoPrint-Enclosure-4.13.2
Done!
```